ieatacid/OctoPrint-TerminalCommands #1: Only first entry is used

EddyMI3d, 2018-01-03:
Hello!

Thanks for that neat plugin!
I just installed version 0.1.0 and added some of the examples you give in the description.
But no matter witch one I click, always the first entry is given out to the printer.

Thanks for help,
Ewald

mane-wt, 2018-01-03:
Hi, I experienced the same problem. /MN

ieatacid, 2018-01-03:
If 0.1.1 doesn't fix it, please provide the OS OctoPrint is running on, the OS and browser you're accessing OctoPrint from and, if you're on Chrome, please `ctrl+shift+i` and provide a [pastebin](https://pastebin.com/) link to the javascript console log output.

EddyMI3d, 2018-01-03:
I just have running a print job, but I will report ASAP

BerndJM, 2018-01-03:
Runing OctoPrint on Raspbian-based "OctoPi" 
accessing with Chrome-Browser no entry works.
Here the console output:   
https://pastebin.com/SQQUJPFq

ieatacid, 2018-01-03:
Ok, I think I got it with 0.1.2.

BerndJM, 2018-01-03:
Indeed you've got it!
Works like a charm.

Just one thought:
why don't use textareas instead input type text for the commands?
It would be much more convenient for complex sequences ...

ieatacid, 2018-01-03:
I used OctoPrint's 'Terminal Filters' source code as a guide and that uses input type text. Is it multi-line input that you're looking for?

BerndJM, 2018-01-03:
Yes, like it is realized e.g. in the customControl-Plugin:
https://github.com/Salandora/octoprint-customControl

But it's not absolutely essential, as I said "just one thought",
because most of the commands will be anyway "one-liners"

EddyMI3d, 2018-01-03:
Runs perfect!

ieatacid, 2018-01-03:
You can set a button to do multiple commands by separating them with a semicolon.

I didn't make the fields multi-line mostly because it would make the settings page look messy.

closed 2018-01-04 by ieatacid

mane-wt, 2018-01-05:
Works perfectly now, thanks.