ieatacid/OctoPrint-TerminalCommands #5: Gcode sequence not executed as configured.

joris57, 2018-02-25:
OctoPrint and OctoPrint-TerminalCommands are both at the latest stable version.

Command sequence as entered in TerminalCommands configuration:
`G0 Z2; G0 X50 Y45 F3000; G0 X50 Y145 F3000; G0 X150 Y145 F3000; G0 X150 Y45 F3000;`

First time execution is ok:

`Send: G0 Z2

Recv: ok

Send: G0 X50 Y45 F3000

Recv: ok

Send: G0 X50 Y145 F3000

Recv: ok

Send: G0 X150 Y145 F3000

Recv: ok

Send: G0 X150 Y45 F3000

Recv: ok`

Second time execution, sequence gets garbled:
`Send: G0 Z2

Recv: ok

Send: G0 X150 Y145 F3000

Recv: ok

Send: G0 X50 Y145 F3000

Recv: ok

Send: G0 X50 Y45 F3000

Recv: ok

Send: G0 X150 Y45 F3000`

joris57, 2018-02-27:
Not solved. Here is the output from 3 consecutive pushes of the button.

Send: G0 Z2
Send: G0 X50 Y45 F1500
Send: G0 X150 Y45 F1500
Send: G0 X150 Y145 F1500
Send: G0 X50 Y145 F1500

Send: G0 Z2
Send: G0 X50 Y45 F1500
Send: G0 X150 Y45 F1500
Send: G0 X50 Y145 F1500
Send: G0 X150 Y145 F1500

Send: G0 Z2
Send: G0 X50 Y145 F1500
Send: G0 X50 Y45 F1500
Send: G0 X150 Y145 F1500
Send: G0 X150 Y45 F1500

ieatacid, 2018-02-27:
Weird. What browser are you using?

joris57, 2018-02-27:
Chrome, but I just tried Firefox and getting the same mixed-up execution. If I push another button in between two executions of the 'offending button', commands are executed in the expected sequence.

PreownedFIN, 2018-02-27:
I just saw the update notification and thought to report that I had also noticed this "issue".

As the issue is alegedly fixed, this doesn't matter that much, but I just ran some of my commands on Firefox (Attached image) and had the issue again. (ON THE OLD VERSION)
[image]

ieatacid, 2018-02-28:
Ok. 0.1.4 should *really* fix this.

joris57, 2018-03-01:
:-) Looks like it's woking correctly now. Thanks for the quick response. This is a really useful plug-in.

joris57, 2019-01-12:
Looks like this problem raised his head again in 0.1.6 ...

Command string as defined in the TerminalCommands plug-in:

M211 S0; G0 X100 Y-25 F2400; M211 S1

Commands as executed in OctoPrint: 

Send: M211 S0
Recv: echo:Soft endstops: Off  Min: X0.00 Y0.00 Z0.00  Max: X290.00 Y290.00 Z260.00
Recv: ok
Send: M211 S1
Recv: echo:Soft endstops: On   Min: X0.00 Y0.00 Z0.00  Max: X290.00 Y290.00 Z260.00
Recv: ok
Send: G0 X100 Y-25 F2400
Recv: ok

joris57, 2019-01-24:
Seems to work again. Don't understand what caused the issue last week.

joris57, 2019-03-28:
Issue is definitely still there. Same button, with no changes, will execute commands in different order from one push to another.

kulfuerst, 2019-04-15:
I can confirm the presence of the issue as well. The order of sent G-Code is not conserved. This issue is present with all command strings I've tested. Below is one example.

Command string as defined in the TerminalCommands plug-in:
M104 S240; M109 S240; G1 E40 F300; G1 E-200 F2000; M300 P500 S1000

Commands as executed in OctoPrint:
Send: M104 S240
Recv: ok
Send: G1 E40 F300
Recv: echo: cold extrusion prevented
Recv: echo: cold extrusion prevented
Recv: ok
Send: G1 E-200 F2000
Recv: echo: cold extrusion prevented
Send: M300 P500 S1000
Recv: echo: cold extrusion prevented
Recv: ok
Recv: ok
Send: M109 S240

Weruminger, 2019-10-30:
Samm Issue with a sequence og G91;G1 Z+10;G90 that will be executed most times 
G91
G90
G1 Z+10

spott, 2020-04-18:
I am also getting random out-of-order execution on semi-colon delineated commands.  It isn't predictable, but sometimes it happens.

Weruminger, 2020-04-18:
I do think it is related to js, because it does not manage arrays in order. i do think about to realize the same funkion as python plugin .... if i will have any time for it ;-)

joris57, 2020-04-26:
I'm closing the issue as it is of no importance anymore for me. If anyone else wants to re-open, please feel free.

closed 2020-04-26 by joris57