ieatacid/OctoPrint-TerminalCommands #6: [FR] Add option to also add buttons to the control tab

PreownedFIN, 2018-02-27:
Hey!

The title for this feature request is pretty self-explanatory.

I just noticed that when setting the commands and when leveling manually, it is quite tedious to jump back and forth between the terminal and control tab. So it would be nice to see the commands under the control tab as well.

If this bothers me too much, I'll probably fork off and submit a PR if and when I get mine working. 😅

ieatacid, 2018-02-27:
Why not just use [Custom Controls](https://plugins.octoprint.org/plugins/customControl/)?

PreownedFIN, 2018-02-27:
Oh!

I didn't know that there was that kind of a plugin as well.

Well, this ended up being a short feature request. 😛

closed 2018-02-27 by PreownedFIN