ieatacid/OctoPrint-TerminalCommands #9: Multiple commands failing checksum and resending indefinitely

TheBum, 2018-06-19:
I attempted to set up the following commands for a Terminal Commands button:

G90 ; G0 X0 Y0 Z10

When I press the button, I see the following in the monitor, repeated over and over:

Send: N12 G90 
 G0 X0 Y0 Z10*36
Recv: Error:No Checksum with line number, Last Line: 11
Recv: Resend: 12
Recv: ok

The printer is stuck in this cycle and the only way to stop it is to power-cycle the printer.

If I send the two commands using separate buttons, they both are accepted without error. The printer is a Snapmaker.

ieatacid, 2018-08-29:
It replaces `;` with `\n`

ieatacid, 2018-12-15:
Try 0.1.6 - it fixes an issue with multiple commands so perhaps it will remedy this as well.