ieatacid/OctoPrint-TerminalCommands #10: M117 (Set LCD Message) not handled correctly

Cerbrus, 2018-12-12:
The terminal commands aren't playing nicely with [`M117`](http://marlinfw.org/docs/gcode/M117.html).

When `M117` is the first command, like this:  
` M117 Reporting setting; M503`  
Then the commands after that aren't executed.

When `M117` appears at the end of a chain of commands:   
`M503; M117 Reporting setting`  
Then the string passed to `M117` is capitalized.

This means that command like these won't work:
`M117 Auto Homing; G28; M117 Auto Homing Done;`
`M117 Centering; G90; G0 Z5; G0 X117.5 Y117.5; G0 Z0; M117 Centered;`

ieatacid, 2018-12-12:
I'll take a look when I get a chance. Probably Saturday at the earliest.

Cerbrus, 2018-12-12:
Thanks for your quick response!

From what I can see, [these lines](https://github.com/ieatacid/OctoPrint-TerminalCommands/blob/master/octoprint_TerminalCommands/static/js/TerminalCommands.js#L48-L53) are probably responsible for capitalizing the string.

It looks like there's a blacklist this command may be missing from?

ieatacid, 2018-12-12:
Weird. I copied that function right from OctoPrint, since it isn't public:

https://github.com/foosel/OctoPrint/blob/master/src/octoprint/static/js/app/viewmodels/terminal.js#L320

ieatacid, 2018-12-12:
Well, OctoPrint only expects one command at a time so multiple may throw an error or produce some weird output.

ieatacid, 2018-12-15:
Try 0.1.6 (your code works for me now)

Cerbrus, 2018-12-15:
I’ll have a look at it this evening 😃

Cerbrus, 2018-12-18:
Sorry, I didn't get around to checking this, yet. Probably this evening.

Cerbrus, 2018-12-18:
Yup! Works like a charm! Thanks for the quick fix!

Especially command like:

`M117 Moving to 0,0; G90; G0 Z5; G0 X0 Y0; G0 Z0; M117 Moved to 0,0`
Work nicely, as the last `M117` only gets logged after the move has finished.

For Auto home / ABL (G28 / G29), the last command executes before the system finished the operation, but I assume that's an OctoPrint limitation.

closed 2018-12-20 by ieatacid