ieatacid/OctoPrint-TerminalCommands #11: [Feature Request] : command field

yet-another-average-joe, 2019-03-27:
Hello,

First of all, thanks for this plugin, that works fine, and does not mess up config.yaml !

when sending a command from a button, it is sent directly to the printer. It would be usefull to first send the command string to the command field, so it could be edited before clicking the "Send" button.

It would add a confirmation level, and the user could edit parameters instead of creating lots of buttons for similar commands. Example : PID autotune for different temperatures.

Even better : some user customizable text could be displayed just below the command field (description, parameters explanations, etc.) when a command button is pushed.

(I don't know if OctoPrint API allows this ; maybe not...)

ieatacid, 2019-03-27:
Have you checked out [Custom Control Editor] (https://plugins.octoprint.org/plugins/customControl)?

yet-another-average-joe, 2019-03-27:
Yes, I did... But it messes up config.yaml. The issue is there : https://github.com/Salandora/octoprint-customControl/issues/22 This issue has never been fixed. Custom Control Editor seems to be an abandonware.

mark-hahn, 2020-01-30:
<cross-posted from octoprint forum>

I would really like to have a terminal command for M851 Z -x.xx where I can plug in x.xx somehow. I tried M851 Z 0 thinking I could scroll up in the terminal history and edit the line. But apparently commands from OctoPrint-TerminalCommands don't end up in the history buffer.
\<snip\>
I was hoping someone could think of a trick, like I tried with a dummy value and editing the line. Now that I think about it, just putting a line in the terminal input line with no return would work. It would only support one option.

I also would kill for multiple lines, another enhancement. I will post a request on the repo.

I'm a programmer by trade. I used python for years, although not recently. Maybe a pull request is in order.