ieatacid/OctoPrint-TerminalCommands #12: Semi-colon no longer works as separator

brendabell, 2019-05-15:
#### Issue you are having

I've been using this plugin on OctoPi for several months. I recently noticed that my terminal commands are no longer working. I have several buttons defined with a series of commands separated by semi-colons.

What's happening is the first command is executed and all other commands generate an error.

For example, I have the following gcode command sequence defined:

`g28 w; m190 s60; m860 s30; m105; g80 n7; g81; g0 x0 y150 z100`

When I click the button, I get the following output:

```Send: G28 W
Send: m190 s60
Send: m860 s30
Send: m105
Send: g80 n7
Send: g81
Send: g0 x0 y150 z100
Recv: tmc2130_home_enter(axes_mask=0x01)
Recv:   0 step=29 mscnt= 479
Recv: tmc2130_goto_step 0 61 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x01
Recv: tmc2130_home_enter(axes_mask=0x02)
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Send: M113 S2
Recv: echo:busy: processing
Recv:   0 step= 1 mscnt=  30
Recv: tmc2130_goto_step 1 33 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: ok
Recv: echo:Unknown command: "m190 s60"(2)
Recv: ok
Recv: echo:Unknown command: "m860 s30"(2)
Recv: ok
Recv: echo:Unknown command: "m105"(2)
Recv: ok
Recv: echo:Unknown command: "g80 n7"(2)
Recv: ok
Recv: echo:Unknown command: "g81"(2)
Recv: ok
Recv: echo:Unknown command: "g0 x0 y150 z100"(2)
Recv: ok```

The first gcode command executes successfully. All others fail with an error.

#### Operating System running OctoPrint

Terminal Commands v0.1.6
OctoPrint v1.3.10
Linux octopi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

#### Printer model & used firmware incl. version

Prusa i3 MK3
Firmware v3.7.0

#### Browser and version of browser, operating system running browser

Chrome 73.0.3683.103
Mac OS X 10.14.3

brendabell, 2019-07-07:
I figured out the issue. If you enter single commands in lower-case it works. Apparently the plug-in is converting to uppercase before the command is sent. If you enter multiple commands in lower case, it only uppercases the first command. Everything after the first semi-colon fails with "unknown command".

hariseldon78, 2019-09-04:
i'm not sure that the issue is only related to uppercase or lowercase commands. 
I used the sequence G91;G1Z-10F2000;G90 , at first all lowercase, then after seeing this issue i converted it to uppercase, but the command randomly didn't get executed correctly, crashing the head in the bed trying to go to absolute -10Z :man_facepalming: 
For now it seems to be solved by using maximum 2 commands in each macro, subdividing the command into his components, and using them in sequence.

vector76, 2020-09-14:
Firing out of order is related to #16 
It appears that javascript client side sends requests but doesn't guarantee the order that they are received by the server.