ieatacid/OctoPrint-TerminalCommands #16: Command do not fire in a conssistent order!

Dracrius, 2020-07-07:
I have experienced this multiple times with my two multi line commands. one for bed leveling the other for estep calibrating. It has randomly decided to auto bed level before homing and it often sends G91 after my G1 F100 E100 which is absolutely useless!

Calibrate E-Steps
G91; G1 F60 E100; G90

Generate BLTouch Mesh
G28; G29

vector76, 2020-09-14:
I am observing this too!  This is a major problem.  On my CNC machine this can be catastrophic.  Extra G4 appears to help somewhat but not completely.

vector76, 2020-09-14:
Ugly hack, probably not ready for prime time, but adding a delay appears to get them to fire in deterministic order:
https://github.com/vector76/OctoPrint-TerminalCommands/commit/bffb50c201fd3f259344134c3f7a55b015315618

ieatacid, 2020-09-14:
@vector76 What about using [setTimeout](https://www.w3schools.com/jsref/met_win_settimeout.asp) instead of sleep

vector76, 2020-09-14:
I think that would work but I would have to create a closure I think.  I'm needing it right now so a quick but non-ideal solution gets me what I need for today.

I would encourage a proper solution, which mine is not.

lscarneiro, 2021-02-16:
Just noticed the same issue, I have a "Purge nozzle" command that does:

```
G91; G1 E15 F200; G1 E-10 F1500; G90
```
So, put in relative mode, extrude 15, retract 10 faster and finally absolute mode, right?

Well, not quite, sometimes the G90 would come before `G1 E-10 F1500` and make my printer retract 25mm (as it "sums"
 the 2 movements) what makes the exact inverse of the proposed sequence.

My "temporary" solution was to use 4 separate commands and live with them but this is definitely a bug!